Hieromon/AutoConnect #573: Compilation error on ESP32

happy12, 2023-01-30:
The compiler errors out with:
 'HTTP_PUT' conflicts with a previous declaration
Under PlatformIO:
[env:esp-wrover-kit]
platform = espressif32
board = esp-wrover-kit
framework = arduino

Hieromon, 2023-01-30:
@happy12 
AutoConnect has completed build testing with CI prior to release. *HTTP_PUT* is an `http_method` enum value declared by the *http_parser.h* of ESP32 SDK, but AutoConnect does not explicitly redefine it. Probably a conflict between **WebServer.h** and **HTTPClient.h**.

However, I will consider improvements to avoid multiple declarations of the `http_method` even if the user sketch indirectly includes the http parser. So could you post the following information about your compilation results?
Refs: https://hieromon.github.io/AutoConnect/faq.html#4-reports-the-issue-to-autoconnect-github-repository

- Version of ESP32 Arduino core
- Compiler Error messages including compile error location
- Dependency graph of PlatformIO
- `#incude` directives part of your sketch code

happy12, 2023-01-31:
maybe it's because I use "ESPAsyncWebServer-esphome" and not "Webserver"?

Hieromon, 2023-01-31:
I don't know. And I have no way of knowing that you were using ESPAsyncWebServer, which was forked to conform to ESPHome.

But it is possible to investigate and determine the cause. Of course you can do it yourself. I need the information commented on [here](https://github.com/Hieromon/AutoConnect/issues/573#issuecomment-1408298801) if I will do an investigation. You say that compilation errors occur, but I have seen no evidence.

I am sure that multiple inclusion of `http_parser.h` is occurring between the libraries you are using. When you include multiple libraries using C++ templates, conflicts between global declarations that do not specify scope are common due to the nature of split compilation. It would not be hard to work around it.

The following SDKs and libraries include `http_parser.h`. You can find out where this include is derived from.

```
tools\sdk\esp32\include\esp_http_server\include\esp_http_server.h:
tools\sdk\esp32\include\esp_http_client\include\esp_http_client.h:
libraries\WebServer\src\HTTP_Method.h:
```

The WebServer library includes `HTTP_Method.h`, which is fine as long as you have not misplaced `WebServer.h` include.

Hieromon, 2023-01-31:
Insufficient information provided makes it difficult to continue the survey. Since this is not an AutoConnect issue, I will move to Discussion and await similar reports from other users.