Hieromon/AutoConnect #584: Error When compiling previously working autoconnect code

Magnus747, 2023-03-02:
Hello, Tried to compile previously working code using autoconnect and get this error: 

.../Documents/Arduino/libraries/AutoConnect/src/AutoConnectElementBasis.h:327:91: error: macro "min" requires 2 arguments, but only 1 given
     : AutoConnectElementBasis(name, "", post), label(String(label)), value(value), min(min), max(max), step(step), magnify(magnify), style(style) {
                                                                                           ^

Cant figure out what is causing the problem. I have updated all my libraries and even deleted all my autoconnect code and i get this error if i #include <AutoConnect.h> at all.

Hieromon, 2023-03-02:
@Magnus747,

> previously working code

Since AutoConnect v1.4.0, the AutoConnect class has been templated and the structure of the header file has changed significantly. Most of the AutoConnect source files will be included in a single sketch file. During this process, `#define` macro directives that were not overloaded in previous versions due to the effect of separate compilation may become overloaded in v1.4.0.

In this case, perhaps there is another library header that includes `#define min` macro forcibly before including `AutoConnect.h`. Please post the `#include` directives in your sketch. I will make a workaround.

Magnus747, 2023-03-03:
So I am using M5Stack libraries and fixed the issue by using their unified library instead of the hardware specific ones. I think they are moving to that, but the workaround may be good if anyone else happens to have old code.

Here are my header files anyway:

#include <M5StickC.h>
#include <EEPROM.h>  
#include <WiFi.h>
#include <WiFiClient.h>
#include <WebServer.h>
#include <AutoConnect.h>
#include <ESPmDNS.h>
#include <AutoConnect.h>
#include <Adafruit_NeoPixel.h>

Hieromon, 2023-03-04:
They seem to have fixed it. In the first place, redefining `min` or `max` by the preprocessor is insane in modern STL C++.
https://github.com/m5stack/M5StickC/commit/9b92d2781a0e3f6a2d8186884066bd2a1bbf981a

closed 2023-03-04 by Hieromon