Hieromon/AutoConnect #589: SPIFFS: mount failed -10025

DitroniX, 2023-04-04:
Have just setup as a quick 'test environment', in order to evaluate AutoConnect as an user option for my SDK boards:

- Arduino IDE 2.0.4
- AutoConnect Library 1.4.2
- HelloWorld example
- ESP32-WROOM-32E-N4
- Module: ESP32 D1 Mini

Probably having a grey moment but note the below unexpected error.  Tried on different boards and the same.

Any thoughts please?

Output
    rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
    configsip: 0, SPIWP:0xee
    clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
    mode:DIO, clock div:1
    load:0x3fff0030,len:1184
    load:0x40078000,len:13220
    ho 0 tail 12 room 4
    load:0x40080400,len:3028
    entry 0x400805e4
    
    **E (1039) SPIFFS: mount failed, -10025**

Many thanks
Dave

wspitts2, 2023-04-04:
You might need to format the spiff folder:
my setupSPIFF returns true if I don't get an error and false if I get an error. I relaunch spiffsSetup after formatting. That seems to do the trick for me.

if(!setupSPIFF()){
    Serial.println("SPIFFS needs to be formatted and trying again");
    SPIFFS.format();
    if(!setupSPIFF()){
      Serial.println("SPIFFS error try #2. Not sure, look into this....");
    }
  }

wspitts2, 2023-04-05:
Well that is more elegant than what I was doing for sure! I guess RTFM a little deeper would have helped.

Hieromon, 2023-04-07:
This is not within the scope of AutoConnect; it is a specification of the ESP32 Arduino Core's SPIFFS wrapper. Please raise this in the ESP32 forum for documentation.

closed 2023-04-07 by Hieromon