Hieromon/AutoConnect #590: Autoconnect Redirect after success

wspitts2, 2023-04-04:
When I connect to the esp32ap through the wifi network list in android. It pops up the configuration screen (great!). I configure everything and then hit connect. Instead of going to the new device IP address /_ac/success it goes to the default 172.217.28.1/_ac/success which is not reachable and barfs a page not found error from the portal popup on android (I believe the SSID is shut down after a successful connection and thus not listening).  Shouldn't there be a default delay between changing the network, and the success page before shutting down the esp32ap SSID SoftAP? If so, how could I implement this.

Hieromon, 2023-04-07:
@wspitts2 
> he default 172.217.28.1/_ac/success which is not reachable and barfs a page not found error from the portal popup on android (I believe the SSID is shut down after a successful connection and thus [...]

No, SoftAP is alive. Stop only DNS if the connection is successful.
Also, the client (your Android device) keeps connection to SoftAP even if the STA side of the ESP32 connects successfully. AutoConnect for ESP32 does not change the network.

> Instead of going to the new device IP address /_ac/success it goes to the default 172.217.28.1/_ac/success which is not reachable and barfs a page not found error from the portal popup on android

This means that the client device (Android) leaves the ESP32's SoftAP and automatically switches connections to the same access point that the ESP32 STA just connected to. Can your android do that?

wspitts2, 2023-04-07:
I will absolutely do that over the next few days! Thanks for the reply!