Hieromon/AutoConnect #591: AutoConnect not trying to connect to all known SSIDs

miroslavpetrov, 2023-04-05:
Hello,

I am experiencing the following issue:

I have saved two SSIDs on the ESP32 and it auto connects only to the last used SSID. If for example I lastly used it with SSID_1 and I turn on the ESP32 near SSID_2(where SSID_1 is not in range) it only tries to connect to SSID_1 and does not try to connect to SSID_2. After that I open the captive portal and select SSID_2 from open ssid menu and it connects as expected. And after that if I again try to connect to SSID_1 when SSID_2 is not in range the issue is the same.

To sum it up: The ESP32 does not try to connect to all known SSIDs(like your smartphone does for example). It only connects to **the last used known SSID**

Hieromon, 2023-04-07:
@miroslavpetrov See the [logic sequence of AutoConnect::begin](https://hieromon.github.io/AutoConnect/lsbegin.html), AutoConnect first executes **WiFi.begin with no arguments**. It tries to connect to the most recent SSID stored by the ESP-IDF SDK. As far as your situation is concerned, AutoConnect is working as expected.

If you want to try to connect to SSID_2 when the connection to SSID_1 fails, the *autoReconnect* setting should help. See the  [`AutoConnectConfig::autoReconnect`](https://hieromon.github.io/AutoConnect/adconnection.html#automatic-reconnect) section of the documentation.

Hieromon, 2023-04-07:
In such cases, finding the AutoConnectConfig settings often results in the expected behavior. However, AutoConnectConfig has many options, and you may need to try several times to achieve the desired configuration.

I move this topic to Discussions because of its potential efficacy in alerting other users.