Hieromon/AutoConnect #593: ESP32 problem

Evgen2, 2023-04-07:
AutoConnect on ESP32 runs in a wrong way.  I take example HelloWorld\HelloWorld.ino on Platformio with platformio.ini 
`[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps = 
	hieromon/AutoConnect@^1.4.2
monitor_speed = 115200
`
As well I uncomment #define AC_DEBUG in \src\AutoConnectDefs.h
This behavior can be better seen in a [short video on YouTube](https://youtu.be/N8XwSMCJckk).
Sometimes all runs well and it is possible to fill  SSID and password for WiFi, sometimes apply not work, but mostly problem with entering in "Configure New AP" and "Open SSIDs" menu item

Check this behavior with Autoconnect 1.3 version -  A behavior is the same.
espressif32 platform version  tested 6.0 and latest 6.1

Hieromon, 2023-04-07:
@Evgen2 The video does not diagnose the detailed operation inside AutoConnect. Please refer to the [documentation](https://hieromon.github.io/AutoConnect/faq.html#4-reports-the-issue-to-autoconnect-github-repository) to collect [`AC_DEBUG`](https://hieromon.github.io/AutoConnect/faq.html#3-turn-on-the-debug-log-options) traces. In some cases, you may also need the PB_DEBUG log.

Evgen2, 2023-04-07:
I was sure everything was working, but deal mainly with esp8266 and now find this on esp-wroom-32

Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)

```
PLATFORM: Espressif 32 (5.2.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20005.220925 (2.0.5)
 - tool-esptoolpy @ 1.40201.0 (4.2.1)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3

``` - 

Now I add PB_DEBUG  and add as well millis() out. Here are two logs - "good"  from reset to show /_ac/config   webpage
and "wronng" from reset to break connect at entering  /_ac/config   webpage

[Good.txt](https://github.com/Hieromon/AutoConnect/files/11181098/Good.txt)
[Wrong.txt](https://github.com/Hieromon/AutoConnect/files/11181099/Wrong.txt)

Hieromon, 2023-04-10:
I'm seeing strange phenomena that I have never seen before.

```
53858 $$$$$ return _token_LIST_SSIDSSID_COUNT HIDDEN_COUNT blk:1270 CONFIG_IP blk:1270 blk:18
```

The log at `$$$$$` prefix is neither AutoConnect nor PageBuilder. However, the string that follows it is clearly from PageBuilder. It appears that the sketch was not built correctly and the constant string area is conflicting.

It seems that the version of platform-espressif32 is out of date (it refers to ESP32 Arduino Core 2.0.5), replace it with the latest Release 6.1.0 and try again.
In addition to that, increase the size of the app partition.

**platformio.ini**

```
board_build.partitions = min_spiffs.csv
```

Evgen2, 2023-04-10:
I have made  many attempts to identify the problem, change  platform versions, add a lot of additional debug message and now find that this stange phenomena associated with the function WiFi.scanNetworks()
Web server kicks client just at execution of WiFi.scanNetworks()

Google finds me similar problem with scanNetworks  and Arduino framework   https://github.com/espressif/arduino-esp32/issues/4554

And seems that I find solution - just call  WiFi.scanNetworks(false, true , true); instead of  WiFi.scanNetworks(false, true);

third parameter -  _bool passive_  what does that mean i have no idea, but it seems work. if it is not working for some time case of  "unstable work"

`int16_t WiFiScanClass::scanNetworks(bool async = false, bool show_hidden = false, bool passive = false, uint32_t max_ms_per_chan = 300U, uint8_t channel = (uint8_t)0U, const char *ssid = (const char *)nullptr, const uint8_t *bssid = (const uint8_t *)nullptr)
`

Hieromon, 2023-04-12:
@Evgen2 I dont know why. The WiFi Driver documentation in the ESP-IDF Programming Guide does indeed describe it. Did you confirm that with my comment mentioning it?

The WiFi.scanNetworks implementation of the ESP32 Arduino core passes WIFI_SCAN_TYPE_PASSIVE;. I cannot confirm anything deeper than that.

https://github.com/espressif/arduino-esp32/blob/224e778b8ed7728a2b104c98c232f34fa34a357c/libraries/WiFi/src/WiFiScan.cpp#L76

Evgen2, 2023-04-15:
It's some sort of black magic.
I tested a number of versions of  _platform = espressif32_ in
[env:esp32dev]
platform = espressif32@version
and have find that with 3.0 all work well,  but with newer versions even pure  WiFi.begin(MySSID, MyPWD); can't connect.
And finally with the help of the guys from the channel https://t.me/ProEsp8266  I have find that my  WiFi router work with pure WPA. After switching to  WPA2  everything is back to normal connecting and scaning

closed 2023-04-15 by Evgen2