Hieromon/AutoConnect #603: AutoConnect on WiFi connection error shows too many redirects.

Pratiksha0102, 2023-05-24:
Hello! Good day to you.

I am currently working with the AutoConnect library, and I have encountered an issue regarding WiFi connection. Interestingly, the library functions properly when connecting to a mobile hotspot. However, when attempting to connect to a router, an error message is displayed. To provide more clarity, I have attached an image of the error below.

[image]

Please let me know if you require any further information or assistance.

Hieromon, 2023-05-27:
Is your router mis-detecting the 302 redirect loop?

Pratiksha0102, 2023-05-29:
I'm sorry, but I'm not familiar with router settings and behavior. Could you please guide me on how to check them? If you need any specific information, I can provide debug logs which you can find here."

> 
[AC] Detected application, connectivitycheck.gstatic.com, 0.0.0.0
[AC] Host:connectivitycheck.gstatic.com,/gen_204,ignored
[AC] Detected application, https://www.google.com, 0.0.0.0
[AC] Host:https://www.google.com,/generate_204,ignored
[AC] Detected application, connectivitycheck.gstatic.com, 0.0.0.0
[AC] Host:connectivitycheck.gstatic.com,/chat,ignored
[AC] Detected application, c.whatsapp.net, 0.0.0.0
[AC] Host:c.whatsapp.net,/_ac/connect,generated:/_ac/connect, allocated
[PB] HTTP_POST /_ac/connect
[AC] Host:172.217.28.1,/_ac/connect,already allocated
[PB] /_ac/connect enable CORS: true
[PB] Chunked, REQ [AC] Queried SSID:Weavers Web 2G AP-1
HEAD CSS_BASE CSS_SPINNER blk:1270 CSS_LUXBAR blk:1270 blk:1270 MENU_PRE blk:1270 MENU_POST CUR_SSID blk:656 
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[AC] WiFi.begin(Weavers Web 2G AP-1,A$ine@2050) ch(6).....................................................................................................timeout
[AC] Host:172.217.28.1,/_ac/result,generated:/_ac/result, allocated
[PB] HTTP_GET /_ac/result
[AC] Host:172.217.28.1,/_ac/result,already allocated
[PB] /_ac/result enable CORS: true
[PB] Chunked, RESULT [AC] Leaves:41[ms]
[AC] Resulting in http://172.217.28.1/_ac/result
[PB] Send canceled
[AC] Quit connecting, status(4)
[AC] Host:172.217.28.1,/chat,ignored
[AC] Detected application, c.whatsapp.net, 0.0.0.0
[AC] Host:c.whatsapp.net,/gen_204,ignored
[AC] Detected application, https://www.google.com, 0.0.0.0
[AC] Host:https://www.google.com,/generate_204,ignored
[AC] Detected application, connectivitycheck.gstatic.com, 0.0.0.0
[AC] Host:connectivitycheck.gstatic.com,/generate_204,ignored
[AC] Detected application, connectivitycheck.gstatic.com, 0.0.0.0
[AC] Host:connectivitycheck.gstatic.com,/chat,ignored
[AC] Detected application, c.whatsapp.net, 0.0.0.0