Hieromon/AutoConnect #606: Failing to store network

fabltd, 2023-05-30:
Updated to 1.4.2 and now AC fails to store network. 

Set network settings:

```
[ 61116][E][WebServer.cpp:649] _handleRequest(): request handler not found
[AC] Detected application, captive.apple.com, 0.0.0.0
[AC] Host:captive.apple.com,/_ac,generated:/_ac, allocated
[AC] Host:172.217.28.1,/_ac,already allocated
[AC] Host:172.217.28.1,/_ac/connect,generated:/_ac/connect, allocated
[AC] Host:172.217.28.1,/_ac/connect,already allocated
[AC] Queried SSID:JRT
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[AC] WiFi.begin(XXX,REPLACED) ch(6)...................................................................................................timeout
[AC] Host:172.217.28.1,proxy-safebrowsing.googleapis.com:443,ignored
[ 95529][E][WebServer.cpp:649] _handleRequest(): request handler not found
[AC] Detected application, proxy-safebrowsing.googleapis.com, 192.168.100.116
[AC] Quit connecting, status(3)
```

Then reboot 

I see the following:

```

[AC] Set persistance
[AC] Current:XXX
[   588][E][Preferences.cpp:50] begin(): nvs_open failed: NOT_FOUND
[AC] Preferences begin failed to import AC_CREDT
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[AC] WiFi.begin() failed
```

So I see the SSID ( Replaced here as XXX) but doesn't connect?

Where do I go to debug from here?

fabltd, 2023-05-30:
I have found the bug.

The inculsion of `#include <SPIFFS.h>` breaks it.

This is used for storage in the main app.

Hieromon, 2023-06-23:
It is not a bug.
See https://hieromon.github.io/AutoConnect/faq.html#nvs_open-failed-not_found-occurs

If the same symptom occurs when the NVS allocation completes, get an AC_DEBUG trace. If you get a trace that the issue is not caused by her NVS allocation, you can reopen this issue. I'll start diagnosing the problem.

So, closed once.

closed 2023-06-23 by Hieromon