Hieromon/AutoConnect #607: AutoConnect not longer runningh on ESP8266

romses, 2023-06-14:
Autoconnect compiles on ESP8266 (Wemos D1 mini), but cannot run.

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v00074040
~ld

I tried several versions and modules.

Hieromon, 2023-06-23:
AutoConnect has been verified to work with Wemons D1 mini. It's probably not an AutoConnect problem.
[Enable AC_DEBUG](https://hieromon.github.io/AutoConnect/faq.html#3-turn-on-the-debug-log-options), then check with the following minimal code.

```cpp
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <AutoConnect.h>

AutoConnect portal;

void setup() {
  delay(1000);
  Serial.begin(115200);
  Serial.println();

  portal.begin();
}

void loop() {
  portal.handleClient();
}
```

JanHBade, 2024-01-11:
same here, at connection to the ESP AP...

any ideas?

at build i have many warnings about
```
 warning: 'DynamicJsonDocument' is deprecated: use JsonDocument instead [-Wdeprecated-declarations]
   66 | using ArduinoJsonBuffer = DynamicJsonDocument;
```

```
User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Unhandled C++ exception: OOM

>>>stack>>>

ctx: cont
sp: 3ffff5d0 end: 3fffffc0 offset: 0000
3ffff5d0:  3fff722c 3fff78fc 00000f59 0000002a
3ffff5e0:  000000fe 00000000 00000000 00000000
3ffff5f0:  00000000 00000000 00000000 00000001  
3ffff600:  00007fff 00000000 3ffe8b74 3fffa30c
3ffff610:  4026cbc2 40218d7c 00000fb0 40226d52
3ffff620:  4026cbc2 40218d7c 00000020 40226d71  
3ffff630:  4026cbc2 3fff78fc 00000fb0 402264c8
3ffff640:  4026cbc2 3fff78fc 00000fb0 40218d7c
3ffff650:  000000f7 3fff7874 3ffff6e0 4023d57c
3ffff660:  00000000 002a002f 00000000 00000000  
3ffff670:  37206d6f 00000000 3fffb00c 0f830f8f
3ffff680:  80000000 00000001 3ffff730 40225776
3ffff690:  3fff8264 3ffff6e0 00000000 00000051  
3ffff6a0:  3fff8264 4026cbc2 4026cbc2 3fff7c30
3ffff6b0:  3ffffd04 00000020 3fffa474 3fff7c34
3ffff6c0:  3ffffd04 3ffff730 3ffff6e0 40218e3a
3ffff6d0:  3ffffd04 3ffff730 3fff4bc4 4021b3ce  
3ffff6e0:  00000000 0046004f 00000000 4021405f
3ffff6f0:  3fff7c58 00000001 3fffa05c 402140a4
3ffff700:  3ffed8fb 3fff7764 3ffff730 3ffffd04  
3ffff710:  0000007b 00000000 3ffff730 4022fb0b
3ffff720:  0000007b 00000000 3fffa05c 402142b7
3ffff730:  3fffaec4 013d013f 80000000 4022fae8  
3ffff740:  3fff7584 0039003f 40215390 4022fae8
3ffff750:  3fff9d0c 000b000f 8a004e45 40227566
3ffff760:  3fffa850 3fffa888 3fffa850 40211cf2
3ffff770:  0000012c 40270930 3ffff780 000004f6 <
3ffff780:  3fffa05c 3fffa9c4 000003d6 40214340
3ffff790:  3ffffd04 3fff8af4 3fff5008 40212235
3ffff7a0:  00000000 3fff8af4 00000020 3fffa9c4  
3ffff7b0:  000004f6 3fff8af4 3fff5008 40214af8
3ffff7c0:  40230e7c 00000000 000003e8 00000000
3ffff7d0:  00000000 3fffa9c4 000004f6 a8b71c01
3ffff7e0:  000004f6 73cc277c 7ae11ce5 de1ad555  
3ffff7f0:  396855a3 2c6d3b15 18685a08 b01b8b0f
3ffff800:  4000b5f9 00000008 00000038 3ffff8e0
3ffff810:  00000008 3ffff8e0 57fc1b1d 131372a9
3ffff820:  b28386fe 5941c1df 8efc2b6a 32910919
3ffff830:  45c808ed d6eea6a2 60bc2832 8fde0238
3ffff840:  2c25f90a 69434f15 9f223f31 851b7cc1
3ffff850:  541cfec0 6b5452d7 ddbab32d 5d67f7d1  
3ffff860:  51b59c71 816f84bc ca0e1eb8 bc2a2c4b
3ffff870:  0a0174ff c019d349 33c08dff 8e705784
3ffff880:  ebfd0fc1 387e935d 279b3922 90fa4494  
3ffff890:  4000b62e c5b4d35b 500d0486 431106f7
3ffff8a0:  4000b6c4 3ffff8e0 3ffffa70 ffffffff
3ffff8b0:  4000b72e 3ffff8c0 00000008 aaf22191
3ffff8c0:  00000000 00000000 4000a333 3ffff9e8  
3ffff8d0:  3ffffa70 3ffff9c8 3fff887c 41a81674
3ffff8e0:  00000000 00000000 00000000 00000000
[... 104 lines cut ...]
3fffff70:  00000050 00000050 3fff7910 00000001  
3fffff80:  3fff4cc4 3fff4d38 feefeffe feefeffe
3fffff90:  feefeffe feefeffe feefeffe 3fff5528
3fffffa0:  3fffdad0 00000000 3fff5514 4022688c  
3fffffb0:  feefeffe feefeffe 3ffe8b70 40101139
<<<stack<<<

last failed alloc call: 40218D7C(4016)

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

last failed alloc caller: 0x40218d7c

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4
tail 4
chksum 0xc9
csum 0xc9
v0007f300
~ld
```