Hieromon/AutoConnect #609: Issue after updating libraries in my PIO project: enableCors()

nostradamux, 2023-09-18:
Hello.
I have been using Autoconnect for a long time in my small temperature sensors, using Wemos D1 mini.
Recently one of those burnt, and I decided to reflash the new Wemos D1 mini.
Then the GIt project that I had that is based on AUTOconnect (version 1.3.1), for flashing, I update all dependencies, and now I receiving error in compilation:

.pio\libdeps\d1_mini\AutoConnect\src\AutoConnectAux.cpp:160:3: error: '**enableCORS**' was not declared in this scope
  160 |   enableCORS(CORS);
      |   ^~~~~~~~~~
.pio\libdeps\d1_mini\AutoConnect\src\AutoConnectAux.cpp: In member function 'bool AutoConnectAux::_load(ArduinoJson::JsonObject&)':
.pio\libdeps\d1_mini\AutoConnect\src\AutoConnectAux.cpp:1047:5: error: '**_cors**' was not declared in this scope; did you mean 'cos'?
 1047 |     _cors = jb[F(AUTOCONNECT_JSON_KEY_CORS)].as<bool>();
      |     ^~~~~
      |     cos
.pio\libdeps\d1_mini\AutoConnect\src\AutoConnectAux.cpp: In member function 'size_t AutoConnectAux::saveElement(Stream&, const std::vector<String>&)':
.pio\libdeps\d1_mini\AutoConnect\src\AutoConnectAux.cpp:1224:11: error: '_cors' was not declared in this scope; did you mean 'cos'

I think this is related with PageBuilder library, but not clear how to solve it.

Here I list versions of packages dependecies in my project:

[image]

Any tip or suggestion to go ahead?
Thanks

tktf50, 2023-12-09:
@nostradamux 

I've run into a similar issue because I'm trying to use an earlier version of the ESP8266 platform in order to try to get "light sleep" working.

It looks like the _enableCORS_ function was added in AutoConnect V1.4.1 (and PageBuilder V1.5.4). You'll need to modify your platformio.ini file to specify PageBuilder V1.5.4 or later if you want to use AutoConnect 1.4.2.

EmbeddedDevver, 2024-06-11:
> @nostradamux
>
> I've run into a similar issue because I'm trying to use an earlier version of the ESP8266 platform in order to try to get "light sleep" working.
>
> It looks like the _enableCORS_ [...]

This was the solution for me as well.
@nostradamux  please close this issue as it is resolved.