Hieromon/AutoConnect #613: AutoConnectPageImpl.hpp:1001:20: error: there are no arguments to 'spi_flash_get_chip_size' that depend on a template parameter, so a declaration of 'spi_flash_get_chip_size' must be available [-fpermissive]

huster-songtao, 2023-11-02:
\AutoConnect\examples\mqttRSSI_FS
open mqttRSSI_FS.ino and build, it throw exceptions:

In file included from d:\Arduino\Sketchbook\libraries\AutoConnect\src/AutoConnectCoreImpl.hpp:14,
                 from d:\Arduino\Sketchbook\libraries\AutoConnect\src/AutoConnectExt.hpp:15,
                 from d:\Arduino\Sketchbook\libraries\AutoConnect\src/AutoConnectExtImpl.hpp:13,
                 from d:\Arduino\Sketchbook\libraries\AutoConnect\src/AutoConnect.h:15,
                 from D:\Github\AutoConnect\examples\mqttRSSI_FS\mqttRSSI_FS.ino:39:
d:\Arduino\Sketchbook\libraries\AutoConnect\src/AutoConnectPageImpl.hpp: In static member function 'static uint32_t AutoConnectCore<T>::_getFlashChipRealSize()':
d:\Arduino\Sketchbook\libraries\AutoConnect\src/AutoConnectPageImpl.hpp:1001:20: error: there are no arguments to 'spi_flash_get_chip_size' that depend on a template parameter, so a declaration of 'spi_flash_get_chip_size' must be available [-fpermissive]
 1001 |   return (uint32_t)spi_flash_get_chip_size();
      |                    ^~~~~~~~~~~~~~~~~~~~~~~
d:\Arduino\Sketchbook\libraries\AutoConnect\src/AutoConnectPageImpl.hpp:1001:20: note: (if you use '-fpermissive', G++ will accept your code, but allowing the use of an undeclared name is deprecated)
D:\Github\AutoConnect\examples\mqttRSSI_FS\mqttRSSI_FS.ino: At global scope:
D:\Github\AutoConnect\examples\mqttRSSI_FS\mqttRSSI_FS.ino:55:69: note: '#pragma message: Warning, LED_BUILTIN is undefined. Assumes Pin #2.'
   55 | #pragma message("Warning, LED_BUILTIN is undefined. Assumes Pin #2.")
      |                                                                     ^
d:\Arduino\Sketchbook\libraries\AutoConnect\src/AutoConnectPageImpl.hpp: In instantiation of 'static uint32_t AutoConnectCore<T>::_getFlashChipRealSize() [with T = AutoConnectConfigExt; uint32_t = long unsigned int]':
d:\Arduino\Sketchbook\libraries\AutoConnect\src/AutoConnectPageImpl.hpp:1242:38:   required from 'String AutoConnectCore<T>::_token_FLASH_SIZE(PageArgument&) [with T = AutoConnectConfigExt]'
d:\Arduino\Sketchbook\libraries\AutoConnect\src/AutoConnectPageImpl.hpp:1647:50:   required from 'PageElement* AutoConnectCore<T>::_setupPage(String&) [with T = AutoConnectConfigExt]'
d:\Arduino\Sketchbook\libraries\AutoConnect\src/AutoConnectCoreImpl.hpp:1447:29:   required from 'bool AutoConnectCore<T>::_classifyHandle(HTTPMethod, String) [with T = AutoConnectConfigExt; HTTPMethod = http_method]'
d:\Arduino\Sketchbook\libraries\AutoConnect\src/AutoConnectCoreImpl.hpp:1093:44:   required from 'void AutoConnectCore<T>::_startWebServer() [with T = AutoConnectConfigExt]'
d:\Arduino\Sketchbook\libraries\AutoConnect\src/AutoConnectCoreImpl.hpp:260:7:   required from 'bool AutoConnectCore<T>::begin(const char*, const char*, long unsigned int) [with T = AutoConnectConfigExt]'
d:\Arduino\Sketchbook\libraries\AutoConnect\src/AutoConnectCoreImpl.hpp:74:15:   required from 'bool AutoConnectCore<T>::begin() [with T = AutoConnectConfigExt]'
D:\Github\AutoConnect\examples\mqttRSSI_FS\mqttRSSI_FS.ino:450:15:   required from here
d:\Arduino\Sketchbook\libraries\AutoConnect\src/AutoConnectPageImpl.hpp:1001:43: error: 'spi_flash_get_chip_size' was not declared in this scope
 1001 |   return (uint32_t)spi_flash_get_chip_size();
      |                    ~~~~~~~~~~~~~~~~~~~~~~~^~

exit status 1

Compilation error: exit status 1

zaid894, 2024-01-24:
I am facing the same issue, and the library is no longer compiling.

johnnyasantoss, 2024-08-16:
Maybe related to https://github.com/espressif/esp-idf/issues/9382#issuecomment-1188497760

jolouis, 2024-12-16:
Not sure whats going on with the maintainer but having accidentally upgraded all the libraries of my build environment the other day I ran into this along with a few of the other noted issues that block compiling.

This one can be fairly easily resolved. As pointed out, the problem is with the ESP-IDF changing the naming and operation of the flash checking method. 
My work around was this:
In src/AutoConnectPageImpl.hpp, update the compiler definitions at the top of the file as follows:
`#if defined(ARDUINO_ARCH_ESP8266)
#include <ESP8266WiFi.h>
extern "C" {
#include <user_interface.h>
}
#elif defined(ARDUINO_ARCH_ESP32)
#ifdef ESP_IDF_VERSION_MAJOR
	#if ESP_IDF_VERSION_MAJOR >= 5
		#include <esp_flash.h>
		#define NEW_ESP_FLASH 1
	#endif
#endif
#ifndef NEW_ESP_FLASH 
	#include <esp_spi_flash.h>
#endif
#include <WiFi.h>
#define ENC_TYPE_NONE WIFI_AUTH_OPEN
#endif`

That will make sure the proper header file is pulled in. Then you need to go down to the actual _getFlashChipRealSize method (its around line 1000 or so) and update it also:
`template<typename T>
uint32_t AutoConnectCore<T>::_getFlashChipRealSize() {
#if defined(ARDUINO_ARCH_ESP8266)
  return ESP.getFlashChipRealSize();
#elif defined(ARDUINO_ARCH_ESP32)
#ifdef NEW_ESP_FLASH
	uint32_t size_flash_chip;
	esp_flash_get_size(NULL, &size_flash_chip);
	return size_flash_chip;
#else
	return (uint32_t)spi_flash_get_chip_size();
#endif
#endif
}`
Maybe somebody with a bit more experience could wrap those changes up into a patch and pull request so everyone wins, but that works well enough for me to allow compiling. In the future if there aren't any updates I may have to switch to a more maintained library instead.