Hieromon/AutoConnect #614: Save data from Custom Page to flash

Yaxit, 2023-11-05:
Hi, as from the title, I'm trying to set up a system with a simple custom page where the user can enter some data (a single number in my case) that persists as the credentials do.
Do you know if this feature exists?
I could only find documentation for the Credentials being saved on flash/nvm, and the JSON format serialization, which is an option but arguably not equally efficient.

Thank you for any indication/suggestion!

f4b0, 2024-06-10:
Hi, have you found a solution for your request because i run in to the same problem?

Thanks

Yaxit, 2024-06-10:
I have not searched further, as this was for a hobby project. However in case you discover anything please post it here