Hieromon/AutoConnect #619: Fresh Install and even the simplest of code will not compile

gnfontaine, 2024-06-07:
I am using a ESP8266 ESP-12 ESP-12F NodeMcu Mini Module and SoftAP works fine without the autoconnect Library.
I tried the simple example and just including the library and errors out with the following code.
Confused 

#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <AutoConnect.h>

ESP8266WebServer server(80);
void setup() {}
void loop() {}

In file included from c:\Users\gnfon\OneDrive\Documents\Arduino\libraries\AutoConnect\src/AutoConnectCore.hpp:31,
                 from c:\Users\gnfon\OneDrive\Documents\Arduino\libraries\AutoConnect\src/AutoConnectCoreImpl.hpp:13,
                 from c:\Users\gnfon\OneDrive\Documents\Arduino\libraries\AutoConnect\src/AutoConnectExt.hpp:15,
                 from c:\Users\gnfon\OneDrive\Documents\Arduino\libraries\AutoConnect\src/AutoConnectExtImpl.hpp:13,
                 from c:\Users\gnfon\OneDrive\Documents\Arduino\libraries\AutoConnect\src/AutoConnect.h:15,
                 from C:\Users\gnfon\OneDrive\Desktop\AutoWifi\AutoWifi.ino:4:
c:\Users\gnfon\OneDrive\Documents\Arduino\libraries\PageBuilder\src/PageBuilder.h:334:74: error: 'HTTPAuthMethod' does not name a type; did you mean 'HTTPMethod'?
  334 |   void  authentication(const char* username, const char* password, const HTTPAuthMethod scheme = HTTPAuthMethod::BASIC_AUTH, const char* realm = NULL, const String& authFail = String(""));
      |                                                                          ^~~~~~~~~~~~~~
      |                                                                          HTTPMethod
c:\Users\gnfon\OneDrive\Documents\Arduino\libraries\PageBuilder\src/PageBuilder.h:367:3: error: 'HTTPAuthMethod' does not name a type; did you mean 'HTTPMethod'?
  367 |   HTTPAuthMethod  _auth;              /**< HTTP authentication scheme */
      |   ^~~~~~~~~~~~~~
      |   HTTPMethod
c:\Users\gnfon\OneDrive\Documents\Arduino\libraries\PageBuilder\src/PageBuilder.h:334:98: error: 'HTTPAuthMethod' has not been declared
  334 |   void  authentication(const char* username, const char* password, const HTTPAuthMethod scheme = HTTPAuthMethod::BASIC_AUTH, const char* realm = NULL, const String& authFail = String(""));
      |                                                                                                  ^~~~~~~~~~~~~~
In file included from c:\Users\gnfon\OneDrive\Documents\Arduino\libraries\AutoConnect\src/AutoConnectCoreImpl.hpp:13,
                 from c:\Users\gnfon\OneDrive\Documents\Arduino\libraries\AutoConnect\src/AutoConnectExt.hpp:15,
                 from c:\Users\gnfon\OneDrive\Documents\Arduino\libraries\AutoConnect\src/AutoConnectExtImpl.hpp:13,
                 from c:\Users\gnfon\OneDrive\Documents\Arduino\libraries\AutoConnect\src/AutoConnect.h:15,
                 from C:\Users\gnfon\OneDrive\Desktop\AutoWifi\AutoWifi.ino:4:
c:\Users\gnfon\OneDrive\Documents\Arduino\libraries\AutoConnect\src/AutoConnectCore.hpp:98:43: error: 'HTTPAuthMethod' does not name a type; did you mean 'HTTPMethod'?
   98 |   void  _authentication(bool allow, const HTTPAuthMethod method);
      |                                           ^~~~~~~~~~~~~~
      |                                           HTTPMethod
In file included from c:\Users\gnfon\OneDrive\Documents\Arduino\libraries\AutoConnect\src/AutoConnectCoreImpl.hpp:14,
                 from c:\Users\gnfon\OneDrive\Documents\Arduino\libraries\AutoConnect\src/AutoConnectExt.hpp:15,
                 from c:\Users\gnfon\OneDrive\Documents\Arduino\libraries\AutoConnect\src/AutoConnectExtImpl.hpp:13,
                 from c:\Users\gnfon\OneDrive\Documents\Arduino\libraries\AutoConnect\src/AutoConnect.h:15,
                 from C:\Users\gnfon\OneDrive\Desktop\AutoWifi\AutoWifi.ino:4:
c:\Users\gnfon\OneDrive\Documents\Arduino\libraries\AutoConnect\src/AutoConnectPageImpl.hpp: In member function 'void AutoConnectCore<T>::_authentication(bool)':
c:\Users\gnfon\OneDrive\Documents\Arduino\libraries\AutoConnect\src/AutoConnectPageImpl.hpp:1835:3: error: 'HTTPAuthMethod' was not declared in this scope; did you mean 'HTTPMethod'?
 1835 |   HTTPAuthMethod  method = _apConfig.auth == AC_AUTH_BASIC ? HTTPAuthMethod::BASIC_AUTH : HTTPAuthMethod::DIGEST_AUTH;
      |   ^~~~~~~~~~~~~~
      |   HTTPMethod
c:\Users\gnfon\OneDrive\Documents\Arduino\libraries\AutoConnect\src/AutoConnectPageImpl.hpp:1836:26: error: 'method' was not declared in this scope
 1836 |   _authentication(allow, method);
      |                          ^~~~~~
c:\Users\gnfon\OneDrive\Documents\Arduino\libraries\AutoConnect\src/AutoConnectPageImpl.hpp: At global scope:
c:\Users\gnfon\OneDrive\Documents\Arduino\libraries\AutoConnect\src/AutoConnectPageImpl.hpp:1848:60: error: 'HTTPAuthMethod' does not name a type; did you mean 'HTTPMethod'?
 1848 | void AutoConnectCore<T>::_authentication(bool allow, const HTTPAuthMethod method) {
      |                                                            ^~~~~~~~~~~~~~
      |                                                            HTTPMethod
C:\Users\gnfon\OneDrive\Desktop\AutoWifi\AutoWifi.ino:7:28: error: cannot declare parameter to be of abstract type 'Server'
    7 | AutoConnect   Portal(Server);
      |                            ^
In file included from C:\Users\gnfon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\ESP8266WiFi\src/WiFiServer.h:31,
                 from C:\Users\gnfon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\ESP8266WiFi\src/ESP8266WiFi.h:40,
                 from C:\Users\gnfon\OneDrive\Desktop\AutoWifi\AutoWifi.ino:1:
C:\Users\gnfon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/Server.h:25:7: note:   because the following virtual functions are pure within 'Server':
   25 | class Server: public Print {
      |       ^~~~~~
In file included from C:\Users\gnfon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/Stream.h:27,
                 from C:\Users\gnfon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/HardwareSerial.h:32,
                 from C:\Users\gnfon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/Arduino.h:303,
                 from C:\Users\gnfon\AppData\Local\Temp\arduino\sketches\76E7C1B873F40DA55ACBB1C0237B1C22\sketch\AutoWifi.ino.cpp:1:
C:\Users\gnfon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/Print.h:55:24: note:     'virtual size_t Print::write(uint8_t)'
   55 |         virtual size_t write(uint8_t) = 0;
      |                        ^~~~~
In file included from C:\Users\gnfon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\ESP8266WiFi\src/WiFiServer.h:31,
                 from C:\Users\gnfon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\ESP8266WiFi\src/ESP8266WiFi.h:40,
                 from C:\Users\gnfon\OneDrive\Desktop\AutoWifi\AutoWifi.ino:1:
C:\Users\gnfon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/Server.h:27:22: note:     'virtual void Server::begin()'
   27 |         virtual void begin() =0;
      |                      ^~~~~
Multiple libraries were found for "ESP8266WebServer.h"
  Used: C:\Users\gnfon\OneDrive\Documents\Arduino\libraries\ESP8266WebServer
  Not used: C:\Users\gnfon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\ESP8266WebServer
Multiple libraries were found for "SD.h"
  Used: C:\Users\gnfon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\SD
  Not used: C:\Users\gnfon\AppData\Local\Arduino15\libraries\SD
exit status 1

Compilation error: cannot declare parameter to be of abstract type 'Server'

gnfontaine, 2024-06-08:
I install older version and works in Arduino Ver 1.8.19 however would be nice if worked in version 2+

JensC, 2024-08-02:
Same with esp32 and Arduino Version: 2.3.3-nightly-20240423
Date: 2024-04-23T03:11:06.602Z
CLI Version: 0.36.0-rc.1

```
/Users/jensi/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-dc859c1e67/esp32s3/include/spi_flash/include/esp_spi_flash.h:6:2: warning: #warning esp_spi_flash.h is deprecated, please use spi_flash_mmap.h instead [-Wcpp]
    6 | #warning esp_spi_flash.h is deprecated, please use spi_flash_mmap.h instead
      |  ^~~~~~~
In file included from /Users/jensi/Documents/Arduino/libraries/AutoConnect/src/AutoConnectCore.hpp:31,
                 from /Users/jensi/Documents/Arduino/libraries/AutoConnect/src/AutoConnectCoreImpl.hpp:13:
/Users/jensi/Documents/Arduino/libraries/PageBuilder/src/PageBuilder.h:301:80: error: 'decltype' cannot resolve address of overloaded function
  301 |     std::is_lvalue_reference<TypeOfArgument<decltype(&RequestHandler::canHandle)>::arg<1>::type>::value,
      |                                                                                ^
/Users/jensi/Documents/Arduino/libraries/PageBuilder/src/PageBuilder.h:301:81: error: template argument 1 is invalid
  301 |     std::is_lvalue_reference<TypeOfArgument<decltype(&RequestHandler::canHandle)>::arg<1>::type>::value,
      |                                                                                 ^
/Users/jensi/Documents/Arduino/libraries/PageBuilder/src/PageBuilder.h:301:89: error: template argument 1 is invalid
  301 |     std::is_lvalue_reference<TypeOfArgument<decltype(&RequestHandler::canHandle)>::arg<1>::type>::value,
      |                                                                                         ^
/Users/jensi/Documents/Arduino/libraries/PageBuilder/src/PageBuilder.h:301:96: error: wrong number of template arguments (1, should be 3)
  301 |     std::is_lvalue_reference<TypeOfArgument<decltype(&RequestHandler::canHandle)>::arg<1>::type>::value,
      |                                     
```

ahmedlamidi, 2024-11-29:
Just need to change the HTTPClient, there is a typo in HTTPUpdate.h, which makes it call the new library instead of the old one. Making that change made it work for me.