Hieromon/AutoConnect #626: Incompatibility ESP32 3.x

valki2, 2024-10-16:
When using V2.0.17 esp32 (by Espressif Systems -> ESP32 Dev Module) the code compiles, when using an upgraded latest esp32 V3.0.5 the library throws errors and wont compile:

In file included from c:\Users\xxx\Documents\Arduino\libraries\AutoConnect\src/AutoConnectCore.hpp:31,
                 from c:\Users\xxx\Documents\Arduino\libraries\AutoConnect\src/AutoConnectCoreImpl.hpp:13,
                 from c:\Users\xxx\Documents\Arduino\libraries\AutoConnect\src/AutoConnectExt.hpp:15,
                 from c:\Users\xxx\Documents\Arduino\libraries\AutoConnect\src/AutoConnectExtImpl.hpp:13,
                 from c:\Users\xxx\Documents\Arduino\libraries\AutoConnect\src/AutoConnect.h:15,
                 from C:\Users\xxx\Documents\Arduino\openAger_lite\openAger_Lite\openAger_Lite.ino:9:
c:\Users\xxx\Documents\Arduino\libraries\PageBuilder\src/PageBuilder.h:301:80: error: 'decltype' cannot resolve address of overloaded function
  301 |     std::is_lvalue_reference<TypeOfArgument<decltype(&RequestHandler::canHandle)>::arg<1>::type>::value,
      |                                                                                ^
c:\Users\xxx\Documents\Arduino\libraries\PageBuilder\src/PageBuilder.h:301:81: error: template argument 1 is invalid
  301 |     std::is_lvalue_reference<TypeOfArgument<decltype(&RequestHandler::canHandle)>::arg<1>::type>::value,
      |                                                                                 ^
c:\Users\xxx\Documents\Arduino\libraries\PageBuilder\src/PageBuilder.h:301:89: error: template argument 1 is invalid
  301 |     std::is_lvalue_reference<TypeOfArgument<decltype(&RequestHandler::canHandle)>::arg<1>::type>::value,
      |                                                                                         ^
c:\Users\xxx\Documents\Arduino\libraries\PageBuilder\src/PageBuilder.h:301:96: error: wrong number of template arguments (1, should be 3)
  301 |     std::is_lvalue_reference<TypeOfArgument<decltype(&RequestHandler::canHandle)>::arg<1>::type>::value,

dncex, 2024-10-23:
I am running in the same issue. It would be great just to know if @Hieromon is going to maintain this lib to make it compatible with arduino core 3.X... (it is just to know, not to push)

Cri4Key, 2025-04-10:
I've created a new pull request (#634) that adds support for ESP32 Arduino Core 3.x. The developer has been inactive for over 2 years, so I don't know if and when the pull request will be examined. In the meantime, you can use the updated versions on my [fork](https://github.com/Cri4Key/AutoConnect). You'll also need updated [PageBuilder](https://github.com/Cri4Key/PageBuilder).

Andrewiski, 2025-04-15:
I patched both AutoConnect and Pagebuilder using the merge above and fix all the errors with ESP32 Core v3.x @Cri4Key thanks for putting in the effort.